joefutrelle/pyifcb #57: ml_analyzed error for bad last line of adc files

hsosik, 2020-04-14:
The MATLAB and Python versions of the ml_analyzed calculation produce an error in some cases where  no inhibit time is available from hdr file AND last line of adc file is bad. Previously we had a special case for ml < 0 (from last adc line), but some cases instead produce ml >> 5 ml (not realistic). I think the better criteria may be to compare the 2nd and 23rd entries on the last line to see if they are more different than a few 10s of milliseconds (the normal diff). In the matlab script just commited I've replaced: 
if ml_analyzed(count) <= 0
with
if abs(adc.Var23(end)-adc.Var2(end)) > 0.1

I haven't fully tested this, but it works for bin  D20180829T144312_IFCB125, which previously gave ml_analyzed = 32 ml, and now gives 3.7348 ml.

joefutrelle, 2020-04-14:
For my reference, [here is the relevant part of the Python code](https://github.com/joefutrelle/pyifcb/blob/b45f4ef89a66c562cccdb417786b266a34e25b03/ifcb/metrics/ml_analyzed.py#L56)

joefutrelle, 2020-04-15:
This is implemented and generates the same result for that bin. I can go ahead and push it now, or if you want to do further testing I can wait for that.

hsosik, 2020-04-21:
I did some more testing and found a need to update the criteria for detecting a bad last line in the adc file. The matlab script is now like this: 
if abs(adc.Var23(end)-adc.Var2(end)) > 0.3

https://github.com/hsosik/ifcb-analysis/blob/master/IFCB_tools/IFCB_volume_analyzed_fromADC.m

joefutrelle, 2020-04-21:
I modified my solution accordingly and created a PR.

https://github.com/joefutrelle/pyifcb/pull/59

joefutrelle, 2020-04-23:
@hsosik are we ready to go on this? If so I will merge the PR and deploy.

hsosik, 2020-05-22:
In the matlab implementation, I am now handling another (rare) special case where multiple lines at the end of the adc file are bad (zero run and inhibit times). [I had a previous commit into ifcb-analysis, that had an incorrect / incomplete implementation of this solution--now I committed what I think is a working version of IFCB_volume_analyzed_fromADC.m]

Can you add this case to the python implementation? In matlab, I'm doing this:
%minor case files with 0 runtime and inhibit time in numerous rows at file end
  if ml_analyzed(count) <= 0 
           runtime = adc.Var2(end-1); %next best info after runtime
           ii = find(adc.Var23);
           modeinhibittime = mode(diff(adc.Var24(ii)));
           inhibittime = adc.Var24(ii(end)) + (size(adc,1)-length(ii)) * modeinhibittime;
           looktime = runtime - inhibittime;
           ml_analyzed(count) = flowrate.*looktime/60;
        end

hsosik, 2020-05-23:
By the way, here is a bin to test the new special case: 
\\sosiknas1\IFCB_data\NESLTER_transect\data\2019\D20190205\D20190205T122609_IFCB127.adc

joefutrelle, 2020-05-26:
What is the correct result for that bin? (so I can check my output)

joefutrelle, 2020-05-26:
Can you briefly describe the algorithm? It looks quite different from the other cases in that it's computing some statistics over the whole file.

hsosik, 2020-05-26:
> What is the correct result for that bin? (so I can check my output)

Correct is a strong word for this case....but my answer is 1.5722 millilters for that bin.

hsosik, 2020-05-26:
> Can you briefly describe the algorithm? It looks quite different from the other cases in that it's computing some statistics over the whole file.

Yes, that's correct. It's like the previous case in getting run time from the end of column 2 of adc, but the inhibittime column has a lot of bad (zero) values in multiple rows at the bottom of the file. So, I'm estimating total inhibittime by taking the last good value (from col 24) and adding an estimate for the rows after that.  I assume each row with missing inhibittime has a value equal to the mode of the measured times (i.e., the "typical" dead time to handle a trigger).

joefutrelle, 2020-05-27:
implemented in #61