joefutrelle/pyifcb #93: Pip installing the latest version does not include the data folder

mike-kaimika, 2025-02-04:
After pip installing the latest version, 1.2, there is a missing `data` folder in site-packages. The other folders `metrics`, `tests` and `viz`, are all there - its just that one folder that's missing. Pip installing the prior version, 1.1.1 includes all for folders:

New release 1.2:
```
root@83ed38177ba8:~# pip install git+https://github.com/joefutrelle/pyifcb@v1.2
root@83ed38177ba8:~# ls /usr/local/lib/python3.11/site-packages/ifcb/
__init__.py  __pycache__  metrics  tests  viz
```

Prior release 1.1.1:
```
root@83ed38177ba8:~# pip install git+https://github.com/joefutrelle/pyifcb@v1.1.1
<output cut for brevity>
root@83ed38177ba8:~# ls /usr/local/lib/python3.11/site-packages/ifcb/
__init__.py  __pycache__  data	metrics  tests	viz
```

The glob in the `pyproject.toml` looks correct (`ifcb/**/*`), so I'm not sure why the data folder is being excluded

shravani-whoi, 2025-02-04:
@mike-kaimika , I am unable to reproduce it on my side. I uninstalled & installed the module explicitly. Could you try uninstalling & reinstalling ?
`(flowenv) (base) shravani.nagala@snagalamacbookpro site-packages % ls -la|grep ifcb
drwxr-xr-x    9 shravani.nagala  staff     288 Feb  4 12:02 ifcb
drwxr-xr-x   10 shravani.nagala  staff     320 Feb  4 12:02 pyifcb-1.2.dist-info
(flowenv) (base) shravani.nagala@snagalamacbookpro site-packages % cd ifcb
(flowenv) (base) shravani.nagala@snagalamacbookpro ifcb % 
(flowenv) (base) shravani.nagala@snagalamacbookpro ifcb % ls -la
total 24
drwxr-xr-x    9 shravani.nagala  staff   288 Feb  4 12:02 .
drwxr-xr-x  111 shravani.nagala  staff  3552 Feb  4 12:02 ..
-rw-r--r--@   1 shravani.nagala  staff  6148 Feb  4 11:56 .DS_Store
-rw-r--r--    1 shravani.nagala  staff    50 Feb  4 12:02 __init__.py
drwxr-xr-x    3 shravani.nagala  staff    96 Feb  4 12:02 __pycache__
drwxr-xr-x   22 shravani.nagala  staff   704 Feb  4 12:02 data
drwxr-xr-x    5 shravani.nagala  staff   160 Feb  4 12:02 metrics
drwxr-xr-x    9 shravani.nagala  staff   288 Feb  4 12:02 tests
drwxr-xr-x    7 shravani.nagala  staff   224 Feb  4 12:02 viz
(flowenv) (base) shravani.nagala@snagalamacbookpro ifcb % cd data
(flowenv) (base) shravani.nagala@snagalamacbookpro data % ls -la
total 240
drwxr-xr-x  22 shravani.nagala  staff    704 Feb  4 12:02 .
drwxr-xr-x   9 shravani.nagala  staff    288 Feb  4 12:02 ..
-rw-r--r--   1 shravani.nagala  staff     79 Feb  4 12:02 __init__.py
drwxr-xr-x  19 shravani.nagala  staff    608 Feb  4 12:02 __pycache__
-rw-r--r--   1 shravani.nagala  staff   6459 Feb  4 12:02 adc.py
-rw-r--r--   1 shravani.nagala  staff   3872 Feb  4 12:02 bins.py
-rw-r--r--   1 shravani.nagala  staff  13874 Feb  4 12:02 files.py
....`

joefutrelle, 2025-02-04:
This is absolutely the .gitignore file. I'll have a PR momentarily.

joefutrelle, 2025-02-04:
Unable to reproduce so far.

joefutrelle, 2025-02-04:
Reproduced in Docker with Python 3.13.

joefutrelle, 2025-02-04:
Confirming that the PR #94 fixes the problem in Docker.

closed 2025-02-04 by joefutrelle