joefutrelle/pyifcb #98: https://pyifcb.readthedocs.io

moi90, 2025-03-09:
I noticed that there is https://pyifcb.readthedocs.io, but without contents. Would you accept a pull request (and provide support) where I set up documentation building using GitHub actions?

joefutrelle, 2025-04-30:
This sounds good to me--I had abandoned that because of an issue with auto-generated documentation for methods with `@lru_cache` decorators just generating the docs for `lru_cache` instead of the decorated method, but if that's no longer an issue then great!

moi90, 2025-05-07:
I don't see any issues when using [Sphinx](https://www.sphinx-doc.org/) 8.2.3 and Python 3.12.3.

moi90, 2025-05-07:
@joefutrelle Could you add me (`moi90`) as a developer to the project at <https://app.readthedocs.org/dashboard/pyifcb/users/>? Then I could try to setup the documentation build.

moi90, 2025-08-13:
@joefutrelle Ping! :)